analogdevicesinc/no-OS #8: Using Code Composer (Texas Instruments) DSP/BIOS and no-OS driver compiler issues

dknowles67, 2015-09-10:
Here is a collection of changes we made to no-OS drivers for use in Code Composer and DSP/BIOS.

File util.c - ERR_PTR function
should return NULL, (or a valid address) - code returns input parameter, cast to pointer, seem odd.

File ad9361_conv.c, post_setup function
3rd parameter in call to function ad9361_dig_tune needs to be cast to type dig_tune_flags.
flags is initialized to zero, which is not defined for type dig_tune_flags.
There are 2 calls to ad9361_dig_tune, which create compilation errors.

File ad9361_api.h
remove trailing comma after last enumeration in ad9361_ensm_mode.

File ad9361_api.c
use dev_err instead of printf globally.
ad9361_get_tx_fir_config - 
    fir_cfg->tx_gain = -6 * (int32_t)(fir_conf & TX_FIR_GAIN_6DB); // typecast to get rid of warning

File ad9361.h
remove trailing comma after last enumeration in:
rx_gain_table_type
rx_gain_table_name
fir_dest
f_agc_target_gain_index_type
rssi_restart_mode
ad9361_pdata_rx_freq
ad9361_pdata_tx_freq
ad9361_clkout
unnamed enumeration ending with LUT_FTDD_ENT
ad9361_clocks
dig_tune_flags
ad9361_bist_mode
debugfs_cmd
unnamed enumeration ending with ID_AD9364

File ad9361.c
Frequent reference to undefined variable "dev", for example:
        dev_err(dev, "Unknown Rx path %"PRIu32, rx_id);
Should be:
        dev_err(&spi->dev, "Unknown Rx path %"PRIu32, rx_id);
?? 
there are dozens of these.  Use of "dev" or "&spi->dev" very inconsistent throughout file.

in ad9361_rx_adc_setup:
    // changed this from original AD driver to get rid of warning
    tmp = (8ULL * scale_snr_1e3 * sqrt_inv_rc_tconst_1e3 *
            min_sqrt_term_1e3) - 50000000ULL;

    // changed this from original AD driver to get rid of warning
    tmp = (invrc_tconst_1e6 >> 1) + 20 * inv_scaled_adc_clk_1e3 *
            data[7] / 80 * 1000ULL;

    // changed this from original AD driver to get rid of warning
    data[12] = min_t(uint32_t, 127U, ((80 * sqrt_inv_rc_tconst_1e3 *
            min_sqrt_term_1e3) - 500000ULL) / 1000000UL);

ad9361_auxdac_get:
// modified to get rid of warning
// can never get here    return 0;
}

ad9361_calculate_rf_clock_chain:
                // modified to get rid of warning
                tmp = (int)(dac_rate / adc_rate) * -1;

ad9361_verify_fir_filter_coef
#ifndef DEBUG // moved this here to get rid of warnings
    return 0;
#else
    struct spi_device *spi = phy->spi;
    uint32_t val, offs = 0, gain = 0, conf, sel, cnt;
    int32_t ret = 0;

 (more unmodified stuff here)

    ad9361_spi_write(spi, REG_TX_FILTER_CONF + offs, conf);

    return ret;
#endif

ad9361_parse_fir:
// strsep not defined in CCS lib
//  while ((line = strsep(&ptr, "\n"))) {
    while ((line = strtok(ptr, "\n"))) {

ad9361_to_refclk_scaler:
    default:
        return -EINVAL;
    }
} // original AD drivers had a semi-colon here, which generates a warning

ad9361_rfpll_int_recalc_rate:
    default:
        return (uint32_t)-EINVAL; // typecast to remove warning in original AD driver
    }

ad9361_clk_register:
    case TX_RFPLL:
        clk->rate = ad9361_rfpll_recalc_rate(clk_priv);
        break;//AD drivers don't have a break here.
    }

rgetz, 2015-09-10:
A fork, with patches, and pull requests is easier to manage, than a list in the tracker.

I think the tracker issue is - for test, we should be compiling with "-ansi -pedantic" (when possible), and clean everything up, not just the things that code composer flagged. When things can't be ansi code (I think the way we initialize structures isn't exactly ansi, since we do it the same way the Linux kernel does), we need to flag that in the comments.

-Robin

buha, 2020-05-15:
This issue is not applicable anymore.

closed 2020-05-15 by buha