analogdevicesinc/no-OS #11: One of the recent driver updates seem to create error message

dknowles67, 2016-01-06:
The change that was called "Implement properly the ad9361_trx_load_enable_fir() function", seems to cause this error report on our system, at this section of ad9361.c (ad9361_validate_enable_fir)
    if (!phy->bypass_tx_fir) {
        max = (tx[DAC_FREQ] / tx[TX_SAMPL_FREQ]) * 16;
        if (phy->tx_fir_ntaps > max) {
            dev_err(dev,
                "%s: Invalid: ratio ADC/2 / TX_SAMPL * 16 > TAPS"
                "(max %"PRIu32", adc %"PRIu32", tx %"PRIu32")",
                __func__, max, rx[ADC_FREQ], tx[TX_SAMPL_FREQ]);
            return -EINVAL;
        }
    }

Furthermore, the dev_err printout should (?) have been something like this:
    if (!phy->bypass_tx_fir) {
        max = (tx[DAC_FREQ] / tx[TX_SAMPL_FREQ]) * 16;
        if (phy->tx_fir_ntaps > max) {
            dev_err(dev,
                "%s: Invalid: ratio DAC / TX_SAMPL * 16 > TAPS"
                "(max %"PRIu32", adc %"PRIu32", tx %"PRIu32" ntaps=%"PRIu32")",
                __func__, max, tx[DAC_FREQ], tx[TX_SAMPL_FREQ],phy->tx_fir_ntaps);
            return -EINVAL;
        }
    }

We were able to get around the problem by moving the phy->filt_valid = true; to the end of the routine
ad9361_trx_load_enable_fir (ad9361_api.c)

dknowles67, 2016-01-06:
We are calling the ad9361_trx_load_enable_fir routine - unlike the sample code.  Not sure if that is what led to the problem.  The sample code calls ad9361_set_tx_fir_config, which sets the values, but does not enable them.

dbogdan, 2016-01-11:
Thanks for telling us about this issue. We have fixed it.
The maximum number of taps was calculated wrong  in some cases (when tx_fir_cfg.tx_path_clks/rx_fir_cfg.rx_path_clks were not defined) because the phy->filt_valid should be set true only when the filters configurations contain tx_path_clks/rx_path_clks definitions.

closed 2016-01-11 by dbogdan