analogdevicesinc/no-OS #29: [PATCH] Update AD9680 JESD base address constant name in fmcadc4.c

michellelloyd, 2017-10-11:
To get the fmcadc4 project to compile as described in https://wiki.analog.com/resources/eval/user-guides/ad-fmcadc4-ebz, I created the following patch.  I hope it is helpful to you.
```
---
 fmcadc4/fmcadc4.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/fmcadc4/fmcadc4.c b/fmcadc4/fmcadc4.c
index 3c9e899..6293b79 100644
--- a/fmcadc4/fmcadc4.c
+++ b/fmcadc4/fmcadc4.c
@@ -225,7 +225,7 @@ int main(void)
 
 #ifdef XILINX
 	ad9680_xcvr.base_address = XPAR_AXI_AD9680_XCVR_BASEADDR;
-	ad9680_jesd.base_address = XPAR_AXI_AD9680_JESD_BASEADDR;
+	ad9680_jesd.base_address = XPAR_AXI_AD9680_JESD_RX_AXI_BASEADDR;
 	ad9680_dma.base_address = XPAR_AXI_AD9680_DMA_BASEADDR;
 	ad9680_0_core.base_address = XPAR_AXI_AD9680_CORE_0_BASEADDR;
 	ad9680_1_core.base_address = XPAR_AXI_AD9680_CORE_1_BASEADDR;
-- 
```

dbogdan, 2017-10-17:
This really depends on what HDL branch is used - we will try to specify for each no-OS branch starting from now on, with what HDL branch is compatible.

michellelloyd, 2017-11-08:
I agree that it would be helpful to have the correlation between HDL and no-OS repositories documented.  However, my patch was specifically so that I can get the fmcadc4.c file to compile, independent of what is in the HDL repository.  I believe this was on the dev branch.

buha, 2020-05-15:
Closing due to inactivity.

closed 2020-05-15 by buha