analogdevicesinc/no-OS #30: GCC: not compilable since that increment on enum is not allowed

alessio31183, 2017-10-13:
Just to focus on this issue: the AD7124.cpp modified for Arduino in extension is not compilable since the enum increment using ++ is not allowed.
I've just modified the code like this:

	/* Initialize registers AD7124_ADC_Control through AD7124_Filter_7. */
	for(regNr = AD7124_Status; (regNr < AD7124_Offset_0) && !(ret < 0); regNr=regNr+1) 
	{
           ..
           ..
        }

Thanks

prashantd2709, 2018-04-04:
Example of driver usage in wiki missing the initialization of structures used. I am stuck at this point. So it will be of great help if someone provides the same.
Thanks

dbogdan, 2018-04-04:
@alessio31183, thanks for let us know.
@prashantd2709, you have open another issue regarding this: https://github.com/analogdevicesinc/no-OS/issues/42

prashantd2709, 2018-04-05:
Hi,
I am trying to use AD7124 no-os driver in embedded c using keil5.
Here regNr=regNr+1 gives the same error as regNr++ viz cannot increment expression of enum type.
Please comment
Thanks

franzASilva, 2018-04-05:
Hi,
In the file AD7124_regs.c change the reg 0x07 value from 0x0044 to 0x0040 - ADC data sheet ON-CHIP REGISTERS / Table 63. Register Summary. Look at this link, too: https://ez.analog.com/message/286889.

Second step, to test, disable "auto" CRC: ad7124_handler->useCRC = 0;

------
Francismar A. Silva

rgetz, 2018-04-05:
Enumerations do not have to be contiguous, 

``enum value {
  a = 0,
  b,
  c = 3 
}``

If I set ``val = b; val++`` all is valid C, but the behavior is undefined. so enum++ is not safe, so it's better for the compiler to complain than generate bad code.

This is a bug that should get removed from this code (and other drivers that do the same).

rather than enum, we could use ``const unsigned int``, or ``const unsigned short`` or ``const unsigned char`` for registers with 32-bit, 16-bit, and 8-bit addresses. The other alternative is some ugly casting to make sure the compiler knows what we are doing, and to put a comment in the header file that these must be in order.

I'm sure others (@larsclausen @mhennerich ?) will have better suggestions.

lclausen-adi, 2018-04-05:
It is valid C, but it is not valid C++, which is why the compiler complains. I'd assume that the project has been configured to use C++.

As a fix I'd just change regNr to unsigned int. That's both valid C and C++. Implicit cast from enum to int works fine in C++, no warnings, no errors.

buha, 2020-05-15:
Closing due to inactivity.

closed 2020-05-15 by buha