analogdevicesinc/no-OS #37: SDK cannot add the libraries of ADV7511

XDtech, 2018-01-31:
When I follow the instructions of the ADV7511 reference design on KC705, I get confused about the step that adding a library into the SDK settings.
1.the library downloaded from Github(no-OS-master/adv7511/library/microblaze/lib/libHDMI_MicroBlazeLib.a) is a compressed file,should I uncompress it first?

2.In the SDK settings ,I add the 'library search path' with the location of above file,and the 'libraries' with 'libHDMI_MicroBlazeLib',then I build the whole project and get an error
 "ld.exe: cannot find -llibHDMI_MicroBlazeLib".
Using the 'libHDMI_MicroBlazeLib.a' as the 'libraries' parameter,the same error occurs(cannot find libHDMI_MicroBlazeLib.a ),even I uncompress the '.a' file and change the 'libraries' with the folder 'libHDMI_MicroBlazeLib',still get wrong.

I am not familiar with the usage of the SDK,and cannot understand why there exist the file,but the compiler says they can't find. 

Env:        win10+vivado2016.4+SDK2016.4
target:    kc705
demo:    hld-2017_r1

any advice will be grateful
/(ㄒoㄒ)/~~

dbogdan, 2018-01-31:
The instructions can be found on the wiki: https://wiki.analog.com/resources/fpga/xilinx/kc705/adv7511

Unfortunately, the libraries were not updated recently, have a look here, too: https://ez.analog.com/message/327173-adv7511-ref-design-no-os-setup-for-zc706

Bamboo1583, 2018-06-17:
I am familiar with your problem.Also,before add the library ,the file "main.c" have a problem(the instructions of the ADV7511 reference design on KC705 also have it),then add the library,no problem.But still cannot find the library libHDMI_ZedBoardLib.a.It show the library is work,but cannot find.
how to deal with ? thanks.

michael1888, 2018-08-28:
have to change name to HDMI_ZynqLib (libHDMI_ZynqLib.a on Github)

buha, 2020-05-15:
This project was moved under `projects/adv7511`. Part of this move is the fact that we can use our new Makefiles build system to generate a complete project with all required fils in `build` directory (generated). Please see this guide on how to build no-OS and feel free to reopen if you still have comments here. 
https://github.com/analogdevicesinc/no-OS/wiki/Building-no-OS-on-Linux
https://github.com/analogdevicesinc/no-OS/wiki/Building-no-OS-on-Windows

closed 2020-05-15 by buha