analogdevicesinc/no-OS #50: clk_div incorrect by factor of 2 spi_engine.c

mxdsgnl, 2018-05-10:
https://wiki.analog.com/resources/fpga/peripherals/spi_engine/instruction_format state:

"...The frequency of the SCLK signal is derived from the module clock frequency using the following formula: f_sclk = f_clk / ((div + 1) * 2). If no prescaler block is present div is 0."

The implementations of spi_engine.c use the following formula:

`clk_div = (spi.spi_ref_clk / msg->spi_clk) - 1;`

The frequency assignment is off by a factor of two.  ... see spi_message->spi_clk.  

`clk_div = (spi.spi_ref_clk / (2 * msg->spi_clk)) - 1;`

matches the wiki and I verified on my target that the sclk frequency matches the spi_message->spi_clk assignment.

lclausen-adi, 2018-05-11:
Hi,

Thanks for the report. We'll fix this.

When we update this we should also change the code so that the divider is rounded up to make sure that the SPI SCLK frequency never exceeds the requested frequency.

stefpopa, 2018-05-11:
Hello,

Thank you for letting us know about this bug. However, our newest projects already use this formula, for example: https://github.com/analogdevicesinc/no-OS/blob/master/ad713x-fmcz/spi_engine.c#L632

In which project did you find this issue?

Thank you!
-Stefan

lclausen-adi, 2018-05-11:
@stefpopa Can you still make sure that we round the divider value up instead of down?

stefpopa, 2018-05-11:
Sure, I will fix this.

buha, 2020-05-15:
This issue seems was not resolved since. However, we're in the process of restructuring no-OS and moving all the projects to the `projects/` folder. In this case, when the `ad5766-sdz` project will be moved, it will use the correct and common `drivers/axi_core/spi_engine/spi_engine.c`. 

Closing this for now, feel free to reopen if you still want to discuss anything here.

closed 2020-05-15 by buha