analogdevicesinc/no-OS #51: config.h file missing for the branch of 2017_r1

sunnysunny3145, 2018-05-14:
I followed the instruction from https://wiki.analog.com/resources/eval/user-guides/ad-fmcdaq2-ebz/software/baremetal 

And try to download all the files I need for the reference design. Since I was using the Vivado 2016.4, I decided to follow the branch of 2017_r1. However, after I got all the files mentioned in the link, the SDK builder told me that there is a file named config.h missing. I tried to use the file here:
https://github.com/analogdevicesinc/no-OS/blob/2017_R1/fmcdaq2/config.h

The SDK shows more errors like:
Description	Resource	Path	Location	Type
expected declaration specifiers or '...' before string constant	config.h	/0604_test1/src	line 70	C/C++ Problem

and lots of errors on 
Description	Resource	Path	Location	Type
unknown type name 'int32_t'	ad9144.c	/0604_test1/src	line 48	C/C++ Problem

Did I pick the correct config.h file for this project? So appreciate if anyone could help me on it.

dbogdan, 2018-05-15:
We will have a look and we will update the wiki documentation according to the latest software changes.

buha, 2020-05-15:
Closing due to inactivity. Feel free to reopen if you still have the issue. Also, you may find a fresh version of the project now moved to `projects/fmcdaq2` which should compile and run as expected.

closed 2020-05-15 by buha