analogdevicesinc/no-OS #52: Possible bug here b/c the channel map adds 1 to the channel number

coryc85, 2018-05-15:
https://github.com/analogdevicesinc/no-OS/blob/02fb97b9958feb39bbc3292d92d363696babdd6e/ad9361/sw/ad9361_api.c#L1216

The ad9361_set_tx_attenuation expects the channel numbers to be 0 or 1.  Speaking specifically about setting Tx attenuation for the TX2 channel (numerical value 1), the ad9361_1rx1tx_channel_map will then take the channel number and add 1 to it, returning a value of 2 to the ad9361_set_tx_attenuation function, But then on line 1216, the returned value is checked against 0 or 1 when setting the attenuation, the problem is that for setting attenuation on TX2, the returned value will be 2 and therefore the attenuation will not actually be set.

dbogdan, 2018-05-15:
Well, you are not allowed to do this in 1RX1TX mode.
You can have RX1TX1, RX2TX2, RX1TX2 and RX2TX1 mappings in 1RX1TX mode (depending on one_rx_one_tx_mode_use_rx_num/one_rx_one_tx_mode_use_tx_num settings), but all the functions should be called exactly as if the RX1TX1 mapping would be used.

coryc85, 2018-05-15:
Thank you for the response.  What exactly is it I am not allowed to do in 1rx1tx mode?  Are you saying that in 1rx1tx mode, I should not be enabling Tx2?  I assumed that 1rx1tx mode meant that only 1 receiver and 1 transmitter is enabled at a time, but I can choose which of them I want.  I simply want to use Tx2 path instead of Tx1 path, and when I set rx1tx1_mode_use_tx_num to 2 and try to set the attenuation, it obviously does not work.  I'm not clear about your statement, all the functions should be called exactly as if the RX1TX1 mapping would be used.

coryc85, 2018-05-15:
Ok thank you for the clarification.  I will see how that works for us and I'll close this issue if that resolves it.  Appreciate the fast response.

coryc85, 2018-05-16:
@dbogdan was correct, when using the part in 1rx1tx mode, and then selecting which of the Tx channels to use (Tx1 or Tx2) by the one_rx_one_tx_mode_use_tx_num setting allows us to always use 0 as the channel number.  Thank you.

closed 2018-05-16 by coryc85