analogdevicesinc/no-OS #159: can't find Makefile.microblaze

abubakr88, 2019-03-27:
Accordding to Build for Microblaze "https://github.com/analogdevicesinc/no-OS/blob/master/ad9361/sw" can't find Makefile.microblaze, what is the solution?

iggy14750, 2019-03-29:
+1 although I am looking for the Zynq Makefile for this part. Does the `platform_xilinx` code work for Zynq? Could I just add the right Makefile around it? Be happy to send a PR. Thanks!

buha, 2020-05-15:
You may see how `projects/adv7511` builds for `kc705` to understand how you could do a no-OS microblaze build.
I also found `scripts/microblaze.mk` which predates this github issue.
Closing this due to being old, feel free to reopen if you still have questions.

closed 2020-05-15 by buha