analogdevicesinc/no-OS #160: Latest AD9371/9375 project

kikd, 2019-03-28:
Hi.

I'm tried to make AD9371/9375 baremetal project for ZC706 board according to [AD9371/AD9375 No-OS Setup](https://wiki.analog.com/resources/eval/user-guides/mykonos/no-os-setup).
However, this repository contains 2 project([no-OS/ad9371](https://github.com/analogdevicesinc/no-OS/tree/master/ad9371) and [no-OS/projects/ad9371/src](https://github.com/analogdevicesinc/no-OS/tree/master/projects/ad9371/src)).

Which project is latest, and which is should to use?

Regards.

mhennerich, 2019-03-28:
@dbogdan can comment more. But the one under projects is going the be the one moving forward, the other is going to be removed soon.

kikd, 2019-04-08:
I understood.
Then obsolete one is removed, is wiki article will updated?

dbogdan, 2019-04-08:
We still have to add an example of configuring the TX DMA in the new project - once this will be available, the old one will be completely removed and the wiki will be updated.

kikd, 2019-04-08:
I got it.
Thanks for your answer.

closed 2019-04-08 by kikd