analogdevicesinc/no-OS #175: AD9361: Compile fails

kylex2, 2019-05-29:
Using

```
$ arm-poky-linux-gnueabi-gcc --version
arm-poky-linux-gnueabi-gcc (GCC) 8.3.0
```

compilation fails with the following error:

```
$ make -f Makefile.linux 
arm-poky-linux-gnueabi-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/opt/poky/2.7/sysroots/cortexa9t2hf-neon-poky-linux-gnueabi -Wall -Werror -Wextra -Iplatform_linux -DLINUX_PLATFORM -DDMA_UIO -Os -ffunction-sections -fdata-sections   -c -o ad9361_api.o ad9361_api.c
arm-poky-linux-gnueabi-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 --sysroot=/opt/poky/2.7/sysroots/cortexa9t2hf-neon-poky-linux-gnueabi -Wall -Werror -Wextra -Iplatform_linux -DLINUX_PLATFORM -DDMA_UIO -Os -ffunction-sections -fdata-sections   -c -o ad9361.o ad9361.c
ad9361.c: In function ‘ad9361_mcs’:
ad9361.c:850:2: error: this statement may fall through [-Werror=implicit-fallthrough=]
  __ad9361_spi_writef(spi, reg, mask, find_first_bit(mask), val)
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ad9361.c:5158:3: note: in expansion of macro ‘ad9361_spi_writef’
   ad9361_spi_writef(phy->spi, REG_ENSM_CONFIG_2,
   ^~~~~~~~~~~~~~~~~
ad9361.c:5160:2: note: here
  case 1:
  ^~~~
ad9361.c: In function ‘ad9361_rfpll_round_rate’:
ad9361.c:6959:6: error: this statement may fall through [-Werror=implicit-fallthrough=]
   if (phy->pdata->use_ext_rx_lo) {
      ^
ad9361.c:6968:2: note: here
  case TX_RFPLL:
  ^~~~
cc1: all warnings being treated as errors
<builtin>: recipe for target 'ad9361.o' failed
make: *** [ad9361.o] Error 1
```

Compilation succeeds if -Werror is removed but are those fall-throughs intentional or not?

amiclaus, 2019-12-09:
Hello,
Please try using `arm-linux-gnueabihf-gcc` when building. It should be error-free.

buha, 2020-05-15:
Closing due to inactivity.

closed 2020-05-15 by buha

kylex2, 2020-05-15:
This seems to have been fixed in your repo with commit e6557f37b314a5657421875e8672d3e4d090efa1. I already fixed it in my local repo and haven't tried yours though.

buha, 2020-05-15:
Thanks @kylex2 for letting me know !