analogdevicesinc/no-OS #222: clock mapping issue in ad9361_clk_register() function

virtualthoughts, 2019-08-23:
I was browsing through the [ad9361.c](https://github.com/analogdevicesinc/no-OS/blob/master/ad9361/sw/ad9361.c) file today and in the ad9361_clk_register() function I noticed that the switch statement mapping wasn't correct.

The offending code is at [here](https://github.com/analogdevicesinc/no-OS/blob/a939258fb0a0777f5a61608184fb1c5a2dc576af/ad9361/sw/ad9361.c#L7126)

I don't know enough about the code base to know if these assignments end up as nop's due to later set operations thus keeping these mappings from being a problem or not.

The mapping as I read it is as follows:

BBPLL_CLK --> BB_REFCLK
ADC_CLK --> BBPLL_CLK
R2_CLK --> ADC_CLK
R1_CLK --> R2_CLK
CLKRF_CLK --> R1_CLK
RX_SAMPL_CLK --> CLKRF_CLK
DAC_CLK --> ADC_CLK
T2_CLK --> DAC_CLK
T1_CLK --> T2_CLK
CLKTF_CLK --> T1_CLK
TX_SAMPL_CLK --> CLKTF_CLK
RX_RFPLL_INT --> RX_REFCLK
TX_RFPLL_INT --> TX_REFCLK

virtualthoughts, 2019-08-23:
After further reading through the code and digesting it I think the intention was to generate each subsequent clock based on the previously calculated clock.  Unfortunately the ordering of these calculations are driven by and outside function, so taken in isolation the switch statement I linked to is confusing.

If this last assessment is accurate please just close out this issue.

buha, 2020-06-05:
Yes, your reasoning is correct, clocks are generated by taking into account the parent rate. The correspondence you provided is actually hierarchical, `child -> parent`. Sorry for super late reply but I'm glad you understood it very quickly. I will close this then.

closed 2020-06-05 by buha