analogdevicesinc/no-OS #247: Memory allocation on every SPI read transaction.

zaporozhets, 2019-09-23:
Usually, no-OS platforms have pure dynamic memory implementation, i.e. malloc/free are slow, danger and could fragment heap. What do you think about to replace heap allocation with a static buffer? I can help with this.

https://github.com/analogdevicesinc/no-OS/blob/baa0c399dacc2659c2d13ea768eeb7da901b2099/projects/ad9361/src/ad9361.c#L588

Cheers,
Taras

commodo, 2019-09-24:
This [generally] makes sense on case-by-case basis.

For the example you have provided, it looks like a static buffer on the stack can be useful.
Feel free to open a PR for this.

Please note that you may need to sign a CLA for the code you provide.
But this is automated one you can do it once you do the PR.

Many thanks :)
Alex

buha, 2020-05-15:
I agree that static buffers are better whenever possible.
There are cases in no-OS where this is done with static buffers on the function stack.
There is also the option for global tx/rx buffers sized so they can fit the largest possible message with that particular chip (taken case-by-case).
We'll keep this in mind and consider hunting down unnecessary malloc's.
Thanks for reporting.
Closing this for now due to inactivity, feel free to reopen if you want to discuss more.

closed 2020-05-15 by buha