analogdevicesinc/no-OS #278: SPI CS may be modified to work correctly

amistad000, 2019-10-17:
the SPI CS address needs to modify to work correctly or failed. I'm not sure if the difference of the hardware. 

in https://github.com/analogdevicesinc/no-OS/blob/e6557f37b314a5657421875e8672d3e4d090efa1/projects/ad9371/src/devices/adi_hal/common.c#L49 SS_DECODE attribute is set. 

so the SPI configuration for AD9528 and AD9371 should be modified 

in 
https://github.com/analogdevicesinc/no-OS/blob/e6557f37b314a5657421875e8672d3e4d090efa1/projects/ad9371/src/app/myk_ad9528init.c#L6-L17

and https://github.com/analogdevicesinc/no-OS/blob/e6557f37b314a5657421875e8672d3e4d090efa1/projects/ad9371/src/app/myk.c#L486-L497

to make noOS driver works. Tx transmits signal, Rx need to be check.

dbogdan, 2019-10-17:
The issue will be fixed today. Sorry for the inconvenience.
Dragos

dbogdan, 2019-10-17:
By the way, did you use the ZC706 carrier? What HDL branch did you use?
I don't see any issue on my side using the current version of the no-OS and hdl master branches.

amistad000, 2019-10-18:
1. zc706 carrier 
2. master checkout on Aug. 14th, 2019. 
3. according to Z-7000 TRM and ILA captured waveform, when SS_DECODE is set, 
the CS and corresponding ss2, ss1, ss0 would be 
0-> {0,0,0} 
1-> {0,0,1}
 and so on. 

I can't figure out why your side is quite different from XILINX TRM.

amistad000, 2019-10-24:
@dbogdan 
set AD9528 CS = 1

[image]

set AD9528 CS = 7
[image]

set AD9371 CS = 6 

[image]

Obviously, the AD9528 with CS configured to 1 work abnormal, and result meets the XILINX ZYNQ-7000 TRM. 

H.A.

buha, 2020-06-05:
Hi @amistad000,
I think the following commits fixed this problem:
43a162c and 0d2fe78

Closing this for now, feel free to reopen if you still have troubles with this.

closed 2020-06-05 by buha