analogdevicesinc/no-OS #302: axi_dac_init in "axi_dac_core.c" only initialize the dual tone in Tx1

amistad000, 2019-10-31:
Hi all, 
the initialization of the dac in master 
https://github.com/analogdevicesinc/no-OS/blob/604da22f181f227df71d57af8d1bb79a972bc3dd/projects/drivers/axi_dac_core/axi_dac_core.c#L855-L864
is different from that in tag "2016_R2" 
https://github.com/analogdevicesinc/no-OS/blob/09740630c2e3f7aaceca1ff913226b443bd06843/ad9371/sw/platform_xilinx/dac_core.c#L479-L501 . 

the code the master is only initialize the four tone in Tx1. 
H.A.

amiclaus, 2019-12-09:
Hello,
We are currently working on adding support also for dual tone. As soon as we have a final version, we'll let you know.
Thank you,
Antoniu

amiclaus, 2020-01-17:
Update: should be fixed with #363

closed 2020-04-13 by amiclaus