analogdevicesinc/no-OS #394: AD9361: Cannot read gain value for split gain table mode (memory access errors)

codypiersall, 2019-12-16:
**What is happening?** I'm attempting to use the split gain table mode of the AD9361, and running into some issues.  I believe I'm using the API correctly and that the implementation of [ad9361_get_split_table_gain][get_gt] is buggy.  This would also apply to the [implementation in the kernel module][kernel-impl], although I have no way to test that.  After compiling this program and the no-OS library with [`-fsanitize=address`][fsan] and reading the gain with [`ad9361_get_rx_rf_gain`][ad9361_get_rx_rf_gain], there are some memory access violations.

The full main.c reproducer is at the bottom of this issue (expand the arrow to see it), but the main gist is that there are memory access violations in the following situation:

1. Compile everything with `-fsanitize=address` (this just *exposes* the buggy memory access, doesn't cause it, of course).
2. Use manual gain control (this is a prerequisite to use the split gain table at all).
3. Use split table mode (specified in the `InitParam` struct).
4. Try to read the gain using [`ad9361_get_rx_rf_gain`](https://github.com/analogdevicesinc/no-OS/blob/9578dc028e70ece512368dee651e6a61f4a4d7f7/ad9361/sw/ad9361_api.h#L373).

**Why is this happening**?  I have a partial answer.  The access violations are happening [when calculating the lmt gain][calc]: the index into `lna_table` is invalid in split gain table mode.  For example, if the RF LO frequency is set to 3.07GHz and I'm in split gain mode, `ad9361_gt(phy)` is going to return `4`.  But the `lna_table` only has a length of 3, so we're reading beyond it.

So I get *how* this is happening, but I don't understand what the whole ad9361_get_split_table_gain function is doing.  I think it should just be doing what `ad9361_get_full_table_gain` function is doing, i.e. just using the `abs_gain_tbl`.  I will send in a PR that does this so we can discuss a concrete change.

[calc]: https://github.com/analogdevicesinc/linux/blob/eadedb6d5a4c235caa1edddca4a209c96698fc3f/drivers/iio/adc/ad9361.c#L1534
[kernel-impl]: https://github.com/analogdevicesinc/linux/blob/eadedb6d5a4c235caa1edddca4a209c96698fc3f/drivers/iio/adc/ad9361.c#L1515-L1550
[fsan]: https://gcc.gnu.org/onlinedocs/gcc/Instrumentation-Options.html
[ad9361_get_rx_rf_gain]: https://github.com/analogdevicesinc/no-OS/blob/9578dc028e70ece512368dee651e6a61f4a4d7f7/ad9361/sw/ad9361_api.h#L373
[get_gt]: https://github.com/analogdevicesinc/no-OS/blob/9578dc028e70ece512368dee651e6a61f4a4d7f7/ad9361/sw/ad9361.c#L1790-L1834

<summary><b>minimal reproducer.</b></summary>
<details>
```c
#include "config.h"
#include "ad9361_api.h"
#include "parameters.h"
#include "platform.h"
#include "adc_core.h"
#include "dac_core.h"

// the ad9361_globals.h includes the phy initialization struct, etc.
// must set manual gain control and split gain table mode!
#include "ad9361_globals.h"

struct ad9361_rf_phy *ad9361_phy;

/***************************************************************************//**
 * @brief main
*******************************************************************************/
int main(void)
{
    // boilerplate init stuff
    default_init_param.gpio_resetb = GPIO_RESET_PIN;
    default_init_param.gpio_sync = -1;
    default_init_param.gpio_cal_sw1 = -1;
    default_init_param.gpio_cal_sw2 = -1;

    gpio_init(default_init_param.gpio_resetb);
    gpio_direction(default_init_param.gpio_resetb, 1);

    spi_init(SPIDEV_DEV, SPI_DEVICE_ID, 1, 0);
    ad9361_init(&ad9361_phy, &default_init_param);
    if (!ad9361_phy) {
        fprintf(stderr, "Could not initialize AD9361!\n");
        exit(1);
    }

    ad9361_set_tx_fir_config(ad9361_phy, tx_fir_config);
    ad9361_set_rx_fir_config(ad9361_phy, rx_fir_config);

    // use gain table or not?
    ad9361_phy->pdata->split_gt = 1;
    // do not skip tuning
    ad9361_phy->pdata->dig_interface_tune_skipmode = 0;
    for (int i = 0; i < 48; i++) {
        int cur_gain;
        int ret = ad9361_set_rx_rf_gain(ad9361_phy, 0, i);
        if (ret) {
            continue;
        }
        ret = ad9361_get_rx_rf_gain(ad9361_phy, 0, &cur_gain);
        if (ret) {
            fprintf(stderr, "bad deal for reading gain %d\n", i);
            continue;
        }
        cur_gain = (int8_t) cur_gain;
        fprintf(stderr, "gain %d, %d\n", i, cur_gain);
    }

    return 0;
}

```
</details>

Thanks for your time.

asmatlock, 2019-12-16:
Hi @codypiersall 

I've experienced issues with the split tables gains as well since some of the later updates to the No-Os model for AD9361.

1.	In function ad9361_get_split_table_gain, the two gain arrays, lna_table and mixer_table, are using an index based on the ad9361_adi_gt_info array, but that array has both full and split table gain info causing the return to be off by 3 since the two gain arrays are based on split table only. We either need to fill the unused indexes of the arrays or adjust the returned index before accessing the arrays.
2.	The gain_db element passed into ad9361_set_rx_gain and the gain returned from ad9361_get_rx_gain are not the same. The set gain function does not expect the LPF gain to be added into the gain_db passed in, but the get rx gain will return the complete gain line up including the present LPF gain.

Hopefully that helps as well.
-Aaron

codypiersall, 2019-12-16:
Hi @asmatlock, and thanks for the feedback.

For 1., I had the fix locally of subtracting 3 from index from `ad9361_gt(phy)`, but in the PR I made (#395), I implemented it similarly to how ad9361_get_full_table_gain is done.

For 2., that's a good point and should probably be addressed.  Maybe in another issue?

dbogdan, 2019-12-17:
Hi, thanks both of you for all the details. We will have a look.

codypiersall, 2019-12-17:
FYI, I just updated my minimal reproducer in the original post to format as code correctly.  That should make it much easier to read.

dbogdan, 2020-10-29:
Fixed by: https://github.com/analogdevicesinc/no-OS/commit/004e8c1d419da913b66a00cda9a710e232ef2ca9

closed 2020-10-29 by dbogdan