analogdevicesinc/no-OS #450: AD9361 init fails with ZCU102

MichalMisiaszek, 2020-01-23:
I pulled HDL_2019.1 repor and this one. Compiled project for zcu102, everything compiled fine. 
I updated memory settings as my board is -05 version (without code will not run at all).
Still it fails to initialize: with message:

ad9361_init : Unsupported PRODUCT_ID 0x0ad9361_init : AD936x initialization errr
Failed to restore state
Failed to restore state

The AD board is connected to HPC0.
Any tips very welcome.

buha, 2020-03-10:
@MichalMisiaszek Could you try updating to latest master ? It looks like an SPI communication issue and there were recent changes that might improve the situation if it's indeed SPI related.

buha, 2020-05-15:
Closing due to inactivity. Feel free to reopen if you still have the issue.

closed 2020-05-15 by buha