analogdevicesinc/no-OS #665: Memory leakage, no need to allocate memory to phy->spi again

l-digital, 2020-06-23:
Line 93-96 of file "no-OS/projects/ad9361/src/ad9361_api.c".

The spi descriptor has been initialized with memory allocated before the function ad9361_init is called. The corresponding spi pointer is stored in the parameter of init_param, and thus phy->spi=init_param->spi (line 97) is sufficient and no need to allocate memory (causing memory leakage) with the pointer assigned to phy->spi before line 97.

Additionaly, line 567 may also need to be reconsidered. It could be better no to release the memory of spi in this functin for consistency since the memory is allocated somewhere else.

amiclaus, 2020-06-23:
> The spi descriptor has been initialized with memory allocated before the function ad9361_init is called. 

Can you please point out where the allocation for the spi descriptor is made before calling ad9361_init?

l-digital, 2020-06-23:
The spi is initialized in the file  no-OS/projects/ad9361/src/main.c, line 534 "status = spi_init(&default_init_param.spi, &spi_param);". The pointer is then store in default_init_param.spi.

amiclaus, 2020-06-26:
Done with #677 

Thank you.

amiclaus, 2020-06-29:
PR merged to master.
Closing this issue.

closed 2020-06-29 by amiclaus