analogdevicesinc/no-OS #681: Need to set proper SPI speed for ad9361

l-digital, 2020-07-03:
In no-OS/projects/ad9361/src/main.c line 465, there is no .max_speed_hz set for spi_param. As the consequence, the functin spi_init_ps (which takes the actual init task) in file no-OS/drivers/platform/xilinx/xilinx_spi.c line 240 will use a default value as the spi clock prescaler (prescaler_default). However, this prescaler_default = XSPIPS_CLK_PRESCALE_64; (line 165 of file no-OS/drivers/platform/xilinx/xilinx_spi.c), which could be too small and cause errors when reading/writing spi. (I observed calibration timeout.) In the earlier code (used by fmcomms2), line 106 of file no-OS/ad9361/sw/platform_xilinx/platform.c, XSPIPS_CLK_PRESCALE_256 is used, and I did not observe any calibration timeout when using the fmcomms2. In order to avoid such spi read/write error cases, a proper .max_spped_hz value should be set for spi_param in main.c, and better set prescaler_default to XSPIPS_CLK_PRESCALE_256 in xilinx_spi.c.

buha, 2020-07-06:
Hi @l-digital ,
Thanks for the feedback,

But I don't think this change is applicable for the following reason:
**AD9361 datasheet lists an SPI timing of 20ns which gives 50MHz max speed. Provided clean electrical SPI lines, the chip would work with a .max_speed_hz = 50000000 value.**

If for you it doesn't work even with the default 64 prescaler (1.5 MHz SPI clock) it means you might have dirty electrical signals on SPI bus. Would be worth checking with an oscilloscope.

I agree that we should have an explicit value for .max_speed_hz in projects/ad9361.
But I don't think the prescaler_default in common code should be changed to anything else than its current value becaue 1-2 MHz are reasonable values for SPI bus.

I suggest you explicitly set .max_speed_hz in your application to whatever value resolves to a prescaler of 256 if that works for your hardware setup.

closed 2020-07-08 by buha

l-digital, 2020-07-11:
Basically I agree with you that the ad9361 chip itself should be able to support higher SPI speed. But I also noticed in the earlier code used by fmcomms2, the 256 prescale value was used. I am using a fmcomms2+zedboard, and the outputs from new code and previous code are quite different. I am not sure if the 256 prescale value was selected purposely there to support fmcomms2+zedboard (means such hardware combination need low speed), or just used the lowest speed without specific reasons. Any insights?

dbogdan, 2020-07-11:
Hi,
I guess that the prescaler was chosen small enough to work for all no-OS projects. I don't know what would be the maximum number for the ZedBoard + FMCOMMS2 setup, but it definitely can be at at least 10 MHz - this is how we configure the Linux setup (https://github.com/analogdevicesinc/linux/blob/master/arch/arm/boot/dts/adi-fmcomms2.dtsi#L20).
Thanks,
Dragos