analogdevicesinc/no-OS #742: What is the device structure?

dawn98f, 2020-09-01:
I have a query, which is regarding the coding part of ad7616.

int32_t ad7616_write(ad7616_dev *dev, uint8_t reg_addr, uint16_t reg_data)

where,

dev - The device structure.

reg_addr - The register address.

reg_data - The register data.

can you explain me what is dev, reg_addr and red_data?

link: -
https://github.com/analogdevicesinc/no-OS/blob/2016_R1/ad7616-sdz/ad7616.c

buha, 2020-09-01:
Hi,
dev - is a ad7616_dev structure that ad7616_setup() creates and allocates.
reg_addr - self explanatory, the register address as in datasheet.
reg_data - the value that you want to write to the register.

dawn98f, 2020-09-01:
thankyou buha but can you give an example for the dev. like in code.

It would be helpful.
Thank you.

mchindri, 2020-09-03:
Hi,
Here is an example:
```
ad7616_dev *dev;
ad7616_setup(ad7616_dev &dev, core, init_param); //This function initializes the dev structure
ad7616_write(dev, 0x1234, 10); //An initialized dev needs to be passed to the write function
```
Hope it is more clear this way.

amiclaus, 2020-09-14:
Marked as resolved.

closed 2020-09-14 by amiclaus