analogdevicesinc/no-OS #843: How to configure and generate a sine wave from AD9361 + Zedboard ?

DatPhungThe, 2020-12-01:
Hello,
I am working with Zedboard and AD9361-FMCOMMS2-EBZ, I want to generate a  continuous sine wave and view it on the oscilloscope. I downloaded HW platform and no-OS software project from Github: (https://github.com/analogdevicesinc/hdl/tree/master/projects/fmcomms2/zed) and (https://github.com/analogdevicesinc/no-OS/tree/master/projects/ad9361).
I built the projects successfully, and run the default mode (DDS) by typing "make run"  in the command line in order to upload bitstream and software into the board.
Now, I want to observe a sine wave from sine_lut[] array values in DMA mode. So I uncommented `#define DAC_DMA_EXAMPLE` . 
But I get the error in the main.c : "

> struct ad9361_rf_phy has no member named tx_dac_init

" at the line: `axi_dac_init(&ad9361_phy->tx_dac, ad9361_phy->tx_dac_init);`  

How to fix above error and generate a sine wave in the osciloscope ?
Please give me some advices ! Thank you very much !

Best Regards,

dbogdan, 2020-12-02:
Hi,
We'll have a look and fix the issue.
Thanks,
Dragos

DatPhungThe, 2020-12-02:
Hi dbogdan,

Thank you very much for your supporting !
BTW, I can run the default mode (DDS) on the FPGA evaluation board. But I see a strange waveform on my oscilloscope. 
What is expected waveform in this case (axi_dac_data_sel = AXI_DAC_DATA_SEL_DDS) ?

Best Regards,
Dat Phung

cristiantpop, 2020-12-02:
Hello,

There where some build problems, they where fixed, please try again with the latest master. 

You could also try to build the project with iio suport:
https://wiki.analog.com/resources/tools-software/no-os-software/iio
and connect the iio-osc as client.

Thank you,
Cristian Pop

DatPhungThe, 2020-12-03:
Hi Cristian,
Thank you for your supporting !
I have just got the latest Github master branch. 
I overcomed build errors. 
But when I run DMA mode (e.g: I uncommented `#define DAC_DMA_EXAMPLE` in the file: _app_config.h_), and observe my Spectrum Analyzer, I realized that  Amplitude Spectrum of Tx wave is very small as the photo:
[](https://drive.google.com/file/d/1ohGQ9WKo-W4gJ8JoGh9NMaVmIB4DYms6/view?usp=sharing)
I think Amplitude should be as large as the case in DDS mode.  This is Spectrum of Tx wave in DDS mode:
      [](https://drive.google.com/file/d/1BvbjSv7vzr-GX6k4qlpVegaTYL2DA4tU/view?usp=sharing)
Could you please explain about this difference ?  Did I miss any configuration ?
Thank you very much !!

Best Regards,
Dat Phung

sathishdanthala, 2021-07-10:
Hii I'm also facing same problem.. The amplitude of the wave is very small in DMA-DDR mode unlike DDS mode... Can anyone help me out??

sathishdanthala, 2021-07-10:
> Hi Cristian,
> Thank you for your supporting !
> I have just got the latest Github master branch.
> I overcomed build errors.
> But when I run DMA mode (e.g: I uncommented `#define DAC_DMA_EXAMPLE` in the [...]

Hiiii Dat Phung...
Did you get any solution regarding this??

adrimbarean, 2021-12-13:
Hello,

This discussion should continue on [EngineerZone](https://ez.analog.com/) if the problems still persist.

Regards,
Andrei

closed 2021-12-13 by adrimbarean

pham-cisteme, 2022-12-13:
@DatPhungThe , could I get your contact to ask you something about this problem? I am looking forward to hear from you! Cam on ban