analogdevicesinc/no-OS #1046: In xilinx_spi.c: XIIC_REPEATED_START macro error?

jmichel3, 2021-06-01:
In xilinx_spi.c, the macro `XIIC_REPEATED_START` is used regardless of whether the type of the IIC descriptor is `IIC_PL` or `IIC_PS`. For example, see line 413 vs line 424 in `xil_i2c_write()`. The Xilinx driver iicps_v3_10, however, defines `XIICPS_REP_START_OPTION`. For the case where the IIC descriptor type is `IIC_PS`, should the correct macro be `XIICPS_REP_START_OPTION`?

buha, 2021-08-20:
Your reasoning is correct, here's the doc for XIicPs_SetOptions():
```
* @param	Options contains the specified options to be set. This is a bit
*		mask where a 1 means to turn the option on. One or more bit
*		values may be contained in the mask. See the bit definitions
*		named XIICPS_*_OPTION in xiicps.h.
```

jmichel3, 2021-08-20:
Awesome, thank you!

closed 2021-08-20 by jmichel3