analogdevicesinc/no-OS #1054: AD9361 1RX/1TX mode

SpontaneousDuck, 2021-06-17:
So I am trying to run my system in 1RX/1TX mode by setting `two_rx_two_tx_mode_enable` in the `AD9361_InitParam` to `false`. I debugged the code and it looks like it goes through and writes to the proper registers to set this. I am using the 2019_r1 HDL image that I built myself to add in some debug hooks. After running the init to disable the second set of channels, the channel enable signals are still high (enable_0 to enable_3 inputs to upack) and thus `upack` is still trying to decode 4 channels from my input stream instead of 2. Is there something else I need to do to disable the second channels? I am only using one channel and don't want to be constantly DMAing empty channels. Thanks for the help!

dbogdan, 2021-06-23:
Hi,
You can update rx_adc_init accordingly (set .num_channels to 2):
https://github.com/analogdevicesinc/no-OS/blob/master/projects/ad9361/src/main.c#L112-L116
We will also do that automatically when two_rx_two_tx_mode_enable is cleared.
Thanks,
Dragos

SpontaneousDuck, 2021-06-25:
So I was able to get this working, but not with 1RX/1TX mode. I have both an FMCOMMS3 and FMCOMMS4 (ad9361 and ad9364). Using the example project in ad9361 mode, I was able to use only two channels. Once I switch the flags to use ad9364 mode https://github.com/analogdevicesinc/no-OS/blob/ac0fe321d58b7305ff8c21384b1a482d6ba6f9b2/projects/ad9361/src/app_config.h#L46 (enabling 1RX/1TX mode), the transfers no longer work properly. It looks like the fifo is filling up too quickly and throttling maybe? 

Here is an image of the AD9364 device working but in 2RX/2TX mode with only 2 channels enabled:
[image]
Here is the same device with all the same settings but running in 1RX/1TX mode also with 2 channels enabled:
[image]

Is something not being configured correctly to cause this? Thanks!

danmois, 2023-04-27:
Hi @SpontaneousDuck ,

Is it possible that the situation you were experiencing be caused by the problem solved in the following PR: https://github.com/analogdevicesinc/no-OS/pull/1745 ?

As you can see there, in the AD9361 project, an invalid rate was selected for axi_dac_core, causing the DMAing of empty channels.

Are you still experiencing the issue when using the master branch for both HDL and no-OS?

Regards,
George

buha, 2023-04-27:
We decided to track all issues in Engineer Zone rather than Github, so I'm closing this here, but feel free to open a thread on EZ if @danmois reply didn't fix the issue.

closed 2023-04-27 by buha