analogdevicesinc/no-OS #1066: I hope to add the no-OS device driver of ad916x series. The official version of ad916x series driver of ADI is too old and has a bug.

gitQqqHs, 2021-08-10:
I hope to add the no-OS device driver of ad916x series. The official version of ad916x series driver of ADI is too old and has a bug.

I found some problems when using the ADI official evaluation board.

There are many bugs in the related projects in ADI's official HDL library, and the given code is modified a lot.

No OS does not have the code related to ad916x series. A separate no OS driver is officially provided, but it is too old and has a bug.

I think the ad916x series code provided separately can be merged into the no OS library, and the old code can also be updated in time.

dbogdan, 2021-08-18:
Hi,
We will add this request to our to-do list.
Thanks,
Dragos

buha, 2022-08-04:
Closing this issue. This topic was added to our to-do list.

closed 2022-08-04 by buha