analogdevicesinc/no-OS #1167: Adxl362 Fifo Configuration Bug

samvangysegem, 2021-12-03:
The current implementation of the Adxl362 features an incorrect way of setting the most significant bit of the fifo watermark level in the fifo registers.  The tested code below corrects this bug. I just wanted to share this since I couldn't create a new branch for a pull request... 

```
void adxl362_fifo_setup(struct adxl362_dev *dev,
			uint8_t  mode,
			uint16_t water_mark_lvl,
			uint8_t  en_temp_read) 
{
	uint8_t write_val = 0;

	write_val = ADXL362_FIFO_CTL_FIFO_MODE(mode) |
              (en_temp_read * ADXL362_FIFO_CTL_FIFO_TEMP) |
              ((0x100 & water_mark_lvl) >> 5);
	adxl362_set_register_value(dev,
				   write_val,
				   ADXL362_REG_FIFO_CTL,
				   1);
	adxl362_set_register_value(dev,
				   water_mark_lvl,
				   ADXL362_REG_FIFO_SAMPLES,
				   1);
}
```

adrimbarean, 2021-12-08:
Hello,

Thank you for submitting this!

Can you make a branch in a fork of the repository and do a pull request from that branch? Alternatively, I can take the code from here and apply it myself. Which do you prefer?

Regards,
Andrei

samvangysegem, 2021-12-08:
Hi Andrei

I followed your steps and made a pull request. 

Kind regards,
Sam

closed 2021-12-10 by amiclaus