analogdevicesinc/no-OS #1228: ADXRS453 Driver compile error on stm32f1xx

GregoryLakewood, 2022-01-28:
It looks like ADXRS453 Driver is not suitable for stm32f1xx since GPIO_TypeDef struct on this ucontroller does not have MODER member. [Context.](https://github-com.translate.goog/Bodmer/TFT_eSPI/issues/589?_x_tr_sl=en&_x_tr_tl=pl&_x_tr_hl=pl&_x_tr_pto=op,sc)

`arm-none-eabi-gcc "../Core/Src/stm32_gpio.c" -mcpu=cortex-m3 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32F103xB -c -I../Core/Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc -I../Drivers/STM32F1xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F1xx/Include -I../Drivers/CMSIS/Include -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -MMD -MP -MF"Core/Src/stm32_gpio.d" -MT"Core/Src/stm32_gpio.o" --specs=nano.specs -mfloat-abi=soft -mthumb -o "Core/Src/stm32_gpio.o"`

`../Core/Src/stm32_gpio.c: In function 'stm32_gpio_get_direction':`

`../Core/Src/stm32_gpio.c:280:29: error: 'GPIO_TypeDef' {aka 'struct <anonymous>'} has no member named 'MODER'; did you mean 'ODR'?
  280 |  *direction = (extra->port->MODER >> (2 * pin)) & 0x3;
      |                             ^~~~~
      |                             ODR`

buha, 2022-02-02:
Hi @GregoryLakewood ,
thanks for reporting this,
let me know if the fix works for you

closed 2022-02-03 by buha

GregoryLakewood, 2022-02-04:
Hey @buha thank you for your work. However I think I made a bad work in making my issue clear.

Firstly, problem is not specific for ADXRS453, but in general stm32f1xx does not have the same `#define` as f4 or other. To be precise it does not have `MODER` nor `GPIO_MODE` defined in the [example header file](https://github.com/dukelec/cdbus_bridge/blob/ca953f815a13525130a6c816561c9d76370d3a44/fw_bl/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal_gpio.h#L132)

I don't know whether me making this issue was smart, at that time I just wanted to point out that no-os stm libraries seem to not be suitable for stm32f1xx uprocessors.

commit 97f71abc (2022-02-02): stm32 gpio: remove MODER access

MODER register seems to not be present on all stm32 mcu's so
avoid accessing it directly.

This commit changes the gpio_get_direction implementation on stm32
by reading the direction configuration set using the initialization
functions. So it doesn't perform a register access.

Closes #1228
PR #1236 (2022-02-03): stm32 gpio: remove MODER access
MODER register seems to not be present on all stm32 mcu's so
avoid accessing it directly.

This commit changes the gpio_get_direction implementation on stm32
by reading the direction configuration set using the initialization
functions. So it doesn't perform a register access.

Closes #1228
commit 4bbad54d (2022-02-03): stm32 gpio: remove MODER access

MODER register seems to not be present on all stm32 mcu's so
avoid accessing it directly.

This commit changes the gpio_get_direction implementation on stm32
by reading the direction configuration set using the initialization
functions. So it doesn't perform a register access.

Closes #1228