analogdevicesinc/no-OS #1254: adrv9001 Linux support

kylex2, 2022-02-14:
Is support for this planned?

adrimbarean, 2022-02-14:
Hello,

[The wiki page](https://wiki.analog.com/resources/eval/user-guides/adrv9001) on this I believe holds the answers. In short, I believe Linux support is available.

If you have more questions please direct them to the [Linux forum on EngineerZone](https://ez.analog.com/linux-software-drivers/) as it has more users and better response time (in general) than this issues tab.

Regards,
Andrei

kylex2, 2022-02-14:
The page your link refers to only mentions how to build no-OS for Intel, Xilinx, STM32 and ADuCM3029. Before opening this issue I tried to build for Linux manually but seems the headless.c file is not suited for Linux. How should I understand this? Is it not supported? Will it be?