analogdevicesinc/no-OS #1436: Problem to build a project

fabien637, 2022-07-12:
I try to build the no-OS ad9361 project on Windows but when I run the make command, an error appears telling me that 'xsct' is "not recognized as an internal or external command , an executable program or a command file."

Can you help me ?

[image]

danmois, 2022-07-27:
Hello fabien,

Did you run the following command: C:\Xilinx\Vitis\2021.1\settings64.bat ?

After running it your build should be fine.

Regards,
George

danmois, 2022-08-01:
Please open a thread at https://ez.analog.com/microcontroller-no-os-drivers/ if the problem is not solved.

Regards,
George

closed 2022-08-01 by danmois