analogdevicesinc/no-OS #1445: Files are missing

willwang1982, 2022-07-25:
I can not find the <drivers/tmr/adi_tmr.h> and <drivers/pwr/adi_pwr.h> which include in no_os_pwm.c . I can not find the folders of tmr and pwr from the website https://github.com/analogdevicesinc/no-OS . So when I build a project about [ad463x](https://github.com/analogdevicesinc/no-OS/tree/master/drivers/adc/ad463x) for ZedBoard,it is always incorrect.  Where  can I get the files ?

danmois, 2022-07-27:
Hello,

Please take a look at the following issue: https://github.com/analogdevicesinc/no-OS/issues/1447. The project seems to be working. 

Are you using the latest version of the master branch?

Regards,
George

wang5227004, 2022-07-27:
It's OK according to #1447. Thank you!

closed 2022-07-27 by danmois