analogdevicesinc/no-OS #1447: Problem about NO-OS Project Build

wang5227004, 2022-07-27:
When I bulided the project of ad463x_fmcz( from https://github.com/analogdevicesinc/no-OS/tree/master/projects/ad463x_fmcz ), there was an error "You do not have sufficient permissions to perform this operation" .  I'm an administrator account. Can you give me some advice?

[image]
[image]

danmois, 2022-07-27:
Hello,

You have to run cmd as administrator.

Regards,
George

closed 2022-07-27 by CiprianRegus

wang5227004, 2022-07-27:
It was the  result that  I runned  cmd as administrator. Except this, any others?

danmois, 2022-07-27:
I have just build successfully the project, in a cmd window run as administrator. It is not sufficient that your Win account is Administrator, you also have to chose "Run as administrator" when opening a cmd window.

wang5227004, 2022-07-27:
You are right. 
Thank you very much!
[image]