analogdevicesinc/no-OS #1470: STM32F4 with AD717x - SPI Communication not Started

dcanrt, 2022-08-16:
I am using STM32F4. I'm struggling to get the library working.When I am debugging, chip select pin setup HIGH. It's working correctly.
[ret = no_os_gpio_direction_output(sdesc->chip_select, NO_OS_GPIO_HIGH);](https://github.com/analogdevicesinc/no-OS/blob/6a503d47eb05b2c0159a907e69cb2917ec8bbba0/drivers/platform/stm32/stm32_spi.c#L83-L97)

But in here it seems that it cannot write to BSRR bit.[`gdesc->port->BSRR` = NO_OS_BIT(sdesc->chip_select->number) << 16;`](https://github.com/analogdevicesinc/no-OS/blob/6a503d47eb05b2c0159a907e69cb2917ec8bbba0/drivers/platform/stm32/stm32_spi.c#L253-L264)

I've tried most things to get it working. I would be glad if you help. @danmois 

'main.c' code.
``` 
  ad717x_dev *my_adc;
  ad717x_init_param adc_init;

  no_os_spi_init_param param;
  stm32_spi_init_param sinit;

  param.chip_select = 4;
  param.device_id = 1;
  param.max_speed_hz = 1000000;
  param.mode = NO_OS_SPI_MODE_0;
  param.bit_order = NO_OS_SPI_BIT_ORDER_MSB_FIRST;
  
  sinit.chip_select_port = GPIOA;
  sinit.get_input_clock = &HAL_RCC_GetPCLK1Freq;

  param.extra = &sinit;
  adc_init.spi_init = param;
  
  adc_init.num_setups = AD717x_MAX_SETUPS;
  adc_init.active_device = ID_AD7172_4;
  adc_init.regs = ad7172_4_regs;
  adc_init.num_regs = sizeof(ad7172_4_regs) / sizeof(ad7172_4_regs[0]);
  
  adc_init.setups[0].bi_unipolar = 0x00;
  adc_init.setups[0].input_buff = 0x00;
  adc_init.setups[0].ref_buff = 0x00;
  adc_init.setups[0].ref_source = EXTERNAL_REF;
  
  adc_init.chan_map[0].channel_enable= 0x01;
  adc_init.chan_map[0].analog_inputs.ainp.pos_analog_input = AIN0;
  adc_init.chan_map[0].analog_inputs.ainp.neg_analog_input = REF_M;
 
  adc_init.chan_map[0].setup_sel = 0x00;  

  
  /*gdesc.extra = &gdesc_extra;
  sdesc->chip_select = &gdesc;
  sdesc = desc->extra;
  desc = my_adc->spi_desc;*/

  /* Initialize the driver instance and let's use the ad7176_2_regs array defined in ad7176_2_regs.h */
  ret = AD717X_Init(&my_adc, adc_init);
    if (ret < 0)
      printf("error");/* Something went wrong, check the value of ret! */
    

  while (1)
  {
      /* Read data from the ADC */
      ret = AD717X_WaitForReady(my_adc, timeout);
 
      ret = AD717X_ReadData(my_adc, &sample);

  }
}
```

buha, 2022-08-17:
Hi @dcanrt ,
I assume you've create a custom .ioc for your purpose.

For spi to work with our stm32 spi platform driver you need to initialize your spi and select the desired MOSI/MISO/SCK as spi signals (pin functions). As for CS setting, choose Software NSS Control. This way, STM32CubeMX will be happy with just the 3 signals above to define a SPI instance.
Separately, define a GPIO Output (GPIOA 4) that you use as Chip Select.

dcanrt, 2022-08-17:
Hi @buha , 
Thank you for your responding. Yes, i used CubeMx and i selected MOSI/MISO/SCK. CubeMX generated code for me.

After than, library initialize Chip Select pin, in here.
according to your no_os_spi_init_param param { .chip_select } on main.c. 

https://github.com/analogdevicesinc/no-OS/blob/6a503d47eb05b2c0159a907e69cb2917ec8bbba0/drivers/platform/stm32/stm32_spi.c#L83-L97

But i said before 
https://github.com/analogdevicesinc/no-OS/blob/6a503d47eb05b2c0159a907e69cb2917ec8bbba0/drivers/platform/stm32/stm32_spi.c#L83-L97
CS pin HIGH in here and work correctly, 

But here it is not working correctly, and so communication not started.
https://github.com/analogdevicesinc/no-OS/blob/6a503d47eb05b2c0159a907e69cb2917ec8bbba0/drivers/platform/stm32/stm32_spi.c#L253-L264

buha, 2022-08-30:
this is interesting. 
what exact board are you trying to run this on ? 
are there any known problems in the errata of your microcontroller ?
did you put an oscilloscope on the CS line to observe what happens ? maybe there is some sort of a pull-up on that line in your electronics

dcanrt, 2022-09-14:
Thank you again. @buha  
I am using STM32F405ZGT . I tried CS pin with logic analyzer. As i said before , it is not working at here:
https://github.com/analogdevicesinc/no-OS/blob/6a503d47eb05b2c0159a907e69cb2917ec8bbba0/drivers/platform/stm32/stm32_spi.c#L253

 When using HAL_GPIO_Write() , this pin working correctly. So there is not some of pull-up on that line. 

Then i have try this. 
``` 
int32_t stm32_spi_write_and_read(struct no_os_spi_desc *desc,
				 uint8_t *data,
				 uint16_t bytes_number)
{
	uint8_t *tx = data;
	uint8_t *rx = data;
	struct stm32_spi_desc *sdesc;
	struct stm32_gpio_desc *gdesc;
	SPI_TypeDef * SPIx;
	if (!desc || !desc->extra || !data)
		return -1;

	if (!bytes_number)
		return 0;

	sdesc = desc->extra;
	gdesc = sdesc->chip_select->extra;
	SPIx = sdesc->hspi.Instance;
        
	//gdesc->port->BSRR = NO_OS_BIT(sdesc->chip_select->number) << 16; //
        HAL_GPIO_WritePin(GPIOA, GPIO_PIN_4 , GPIO_PIN_RESET);
        SPI1->CR1 |= SPI_CR1_SPE;
	//__HAL_SPI_ENABLE(&sdesc->hspi);
	while(bytes_number--) {
        	while(!(SPI1->SR & SPI_SR_TXE));
                
		*(volatile uint8_t *)&SPI1->DR = *tx++;
		while(!(SPI1->SR & SPI_SR_RXNE))
			;
		*rx++ = *(volatile uint8_t *)&SPI1->DR;
	}
	//__HAL_SPI_DISABLE(&sdesc->hspi);
	//gdesc->port->BSRR = NO_OS_BIT(sdesc->chip_select->number); //HAL_GPIO_WritePin(GPIOA, NO_OS_BIT(4), GPIO_PIN_SET); 
        HAL_GPIO_WritePin(GPIOA, GPIO_PIN_4 , GPIO_PIN_SET);
	return 0;
}
```
This time, data started to come from SPI. But when it comes to this function, it cannot pass L394 and gets HARD FAULT error.
https://github.com/analogdevicesinc/no-OS/blob/e5bebb8c75a39358baab54000d0ebb59f7b26479/drivers/adc/ad717x/ad717x.c#L388-L405

I think there is an error with pointers or addresses.

buha, 2022-10-14:
Sorry for late reply, we are mostly active on Engineer Zone...
Changing BSRR reg access with HAL_GPIO_WritePin is correct. Actually, I initially used HAL_GPIO_WritePin in this code but changed to using BSRR due to much better performance.
I don't see how your change could make that AD717X_GetReg() fail, I'd need to be able to replicate your setup and use a debugger for that.

realathu, 2022-12-06:
Hi! @dcanrt , Did you get this problem sorted out?....I'm experiencing the same issue!... My set up is AD7768-1 + (Nucleo-144 (F767ZI ) | Discovery (F407VGt))

amiclaus, 2023-04-11:
Closing issue. For further support, please reach our support forum: https://ez.analog.com/microcontroller-no-os-drivers

closed 2023-04-11 by amiclaus