analogdevicesinc/no-OS #1487: Can you add a sample project code for ad717x?

lxguang231, 2022-09-04:
Can you add a sample project code for ad717x, multi-channel application?

sky-akash05, 2022-10-03:
hi, I will love to contribute to your project . Will feel grateful i you can assign this to me,i can add a sample project code for ad717x.

mphalke, 2022-10-03:
Hello, there are few examples we have created for AD717x family of devices (using Mbed platform). Here are the wiki page and software links:
https://wiki.analog.com/resources/tools-software/product-support-software/ad717x_ad411x_mbed_example
https://wiki.analog.com/resources/tools-software/product-support-software/ad717x_mbed_iio_support

If you are specifically looking for STM32 platform, then code can be ported to STM32 with some efforts. The platform drivers needs to be ported from Mbed to STM32. They are available below:
https://github.com/analogdevicesinc/no-OS/tree/master/drivers/platform

Below article can help you to port the code from one platform to another:
https://www.analog.com/en/analog-dialogue/articles/understanding-and-using-the-no-os-and-platform-drivers.html

danmois, 2022-11-04:
Hello,

A sample project for the ad717x no-OS project is not currently planned. You can see more information about the driver and the use of the API at the following address: https://wiki.analog.com/resources/tools-software/uc-drivers/ad717x.

Regards,
George