analogdevicesinc/no-OS #1598: AGC set bug

FullyArticulate, 2023-01-10:
In drivers/rf-transceiver/navassa/devices/adrv9001/public/src/adi_adrv9001_rx_gaincontrol.c:320
```
for (regId = 0; regId < REGISTER_ID; regId++) {
        ADI_EXPECT(adrv9001_AnalogRxMemMap_OrxBlockDetUlbth_Set, device, anaAddr, regId, agcCfg->peak.apdHighThresh);
        ADI_EXPECT(adrv9001_AnalogRxMemMap_OrxBlockDetLlbth_Set, device, anaAddr, regId, agcCfg->peak.apdLowThresh);
}
```

The first (...OrxBlockDetUlbth_Set) translates as:
  `status = adi_bf_hal_Field_Write(device, (instance + 0x70 + 0x0 + (regInstance * 0x1)), (value >> 0), 0x3f, 0x0);`
The second (...OrxBlockDetLlbth_Set) translates as:
  `status = adi_bf_hal_Field_Write(device, (instance + 0x69 + 0x0 + (regInstance * 0x1)), (value >> 0), 0x3f, 0x0);`

In other words, this loop keeps overwriting the previous values:
regId = 0:
   Write to 70 + 0 + 0 and 69 + 0 + 0
regId = 1
   Write to 70 + 0 + 1 and 69 + 0 + 1 (aka 70, which was just written in regId = 0)

Without further docs, I can't diagnose what this is supposed to be, but this is almost certainly wrong.

tfcollins, 2023-01-10:
@FullyArticulate this question is specific to the transceiver API itself which is supported by a specific group. Can you post your question here: https://ez.analog.com/wide-band-rf-transceivers/design-support-adrv9001-adrv9007/

FullyArticulate, 2023-01-10:
Done. [https://ez.analog.com/wide-band-rf-transceivers/design-support-adrv9001-adrv9007/f/q-a/566177/bug-adrv9001-agc-settings-are-likely-incorrect](url)

tfcollins, 2023-01-11:
Closing. EZ will take over.

closed 2023-01-11 by tfcollins