analogdevicesinc/no-OS #1652: Bug in AD9361 register collection for fast lock

smunaut, 2023-02-16:
The code has :
```c
        x = ad9361_spi_readf(spi, REG_RX_VCO_BIAS_1 + offs, VCO_BIAS_TCF(~0));
        y = ad9361_spi_readf(spi, REG_RX_CP_CURRENT + offs, CHARGE_PUMP_CURRENT(~0));
        /* Wide BW option: N = 1
        * Set init and steady state values to the same - let user space handle it
        */
        val[6] = (x << 3) | y;
```

But even though the datasheet does say "left shift by 3", this is because the field being referened ( `VCO_BIAS_TCF` ) is not starting at bit 0. But the function `ad9361_spi_readf` will read it then shift it so that it starts at bit 0. And then to position it properly for fast lock, you need to shift it by 6 and not by 3 so it doesn't overlap with the other field `CHARGE_PUMP_CURRENT` which is 6 bits wide.

amiclaus, 2023-03-27:
Hello, thanks for pointing that out. We'll have a fix soon.

amiclaus, 2023-03-30:
Fixed with #1717 . Issue closed.