analogdevicesinc/no-OS #1973: Supported Samples per Converter per Frame (S)

GbGp, 2023-10-23:
Hello, I noticed that the "samples_per_frame" parameter (JEDEC204 "S") is harcoded to '1' in `axi_jesd204_tx.c`:
https://github.com/analogdevicesinc/no-OS/blob/master/drivers/axi_core/jesd204/axi_jesd204_tx.c#L883C5-L883C37

On the other way there are some devices where a value of 2 and 3 can be used (see https://github.com/analogdevicesinc/hdl/blob/master/projects/dac_fmc_ebz/common/config.tcl).
Is this mode not supported in this driver?

danmois, 2024-01-15:
Hello,

The "samples_per_frame" parameter determines the "BYTES_PER_FRAME" or "octets_per frame" (F) parameter:
https://github.com/analogdevicesinc/hdl/blob/d4f33da416866799f7ed310547669bd36e21bb35/library/jesd204/ad_ip_jesd204_tpl_adc/ad_ip_jesd204_tpl_adc.v#L116

In the no-OS JESD projects, this F parameter is given in the project configuration section:
e.g., https://github.com/analogdevicesinc/no-OS/blob/0c5f60d239d947658ee170eecf2cf075fe48c686/projects/fmcdaq2/src/app/fmcdaq2.c#L433C3-L433C25
or
e.g. https://github.com/analogdevicesinc/no-OS/blob/0c5f60d239d947658ee170eecf2cf075fe48c686/projects/ad9081/src/app_config.h#L62

Until now, only use cases with S = 1 were tested, although the HDL support other values for this parameter.

For further questions, please open a thread on https://ez.analog.com/microcontroller-no-os-drivers/.  

Regards,
George

amiclaus, 2024-02-13:
Closing this. For further assistance please open a thread on https://ez.analog.com/microcontroller-no-os-drivers/ .

closed 2024-02-13 by amiclaus