analogdevicesinc/no-OS #1992: adxl345_set_activity_detection

Mark-Walen, 2023-11-05:
In function void adxl345_set_activity_detection(adxl345_t *adxl345,
									uint8_t act_on_off,
									uint8_t act_axes,
									uint8_t act_ac_dc,
									uint8_t act_thresh,
									uint8_t act_int);
old_act_inact_ctl = adxl345_get_register_value(adxl345, ADXL345_INT_ENABLE);
should it be:
old_act_inact_ctl = adxl345_get_register_value(adxl345, ADXL345_ACT_INACT_CTL);

rbolboac, 2023-11-07:
Hello @Mark-Walen ,

Thank you for reporting this.
A PR has been created with the fix you mentioned.

Best Regards,
Ramona

closed 2023-11-07 by rbolboac

Mark-Walen, 2023-11-08:
Hello [rbolboac](https://github.com/rbolboac),
 Actually there is same issue in [void adxl345_set_inactivity_detection](https://github.com/analogdevicesinc/no-OS/blob/master/drivers/accel/adxl345/adxl345.c#L473).

rbolboac, 2023-11-08:
Hello @Mark-Walen ,

Thank you for pointing this out as well!
A new PR has been created: https://github.com/analogdevicesinc/no-OS/pull/1999

Best Regards,
Ramona