analogdevicesinc/no-OS #2041: Questions on the use of web services

yyllxxhh, 2023-12-06:
In the **iio_app_run** function, there is a continuous execution of **iio_step** to respond to requests from the upper computer.

If network services are enabled, the program will enter the **accept_network_clients** function.

```bash
#if defined(NO_OS_NETWORKING) || defined(NO_OS_LWIP_NETWORKING)
	if (desc->server) {
		ret = accept_network_clients(desc);
		if (NO_OS_IS_ERR_VALUE(ret) && ret != -EAGAIN)
			return ret;
```

However, in the **accept_network_clients** function, the **while(true)** loop causes the program to be blocked, preventing it from entering iiod_conn_step to handle requests from the upper computer.

```bash
#if defined(NO_OS_NETWORKING) || defined(NO_OS_LWIP_NETWORKING)

static int32_t accept_network_clients(struct iio_desc *desc)
{
	struct tcp_socket_desc *sock;
	struct iiod_conn_data data;
	int32_t ret;
	do {
		uint32_t id;
		ret = socket_accept(desc->server, &sock);
		if (NO_OS_IS_ERR_VALUE(ret))
			return ret;

		data.conn = sock;
		data.buf = no_os_calloc(1, IIOD_CONN_BUFFER_SIZE);
		data.len = IIOD_CONN_BUFFER_SIZE;

		ret = iiod_conn_add(desc->iiod, &data, &id);
		if (NO_OS_IS_ERR_VALUE(ret))
			return ret;

		ret = _push_conn(desc, id);
		if (NO_OS_IS_ERR_VALUE(ret))
			return ret;
	} while (true);

	return 0;
}
#endif
```

CiprianRegus, 2023-12-06:
Hi @yyllxxhh,

Can you provide more details on what your hardware setup is and what target are you running IIOD on? Is it Linux userspace or an MCU with WiFi / ADIN1110?

yyllxxhh, 2023-12-06:
Hi @CiprianRegus,

Certainly. In the context of the hardware setup I described earlier, I am running IIOD on the MCU, specifically the STM32F103ZET6. The IIOD is operating on the MCU itself, not in a Linux userspace or on a separate device. The connection is established through the W5500 Ethernet chip using the SOCKET API functions provided by WIZNET. The ADC chip AD7606 is responsible for analog-to-digital conversion within the system. The host machine, running the Windows 11 operating system, utilizes the IIO OSC client to interface with the IIOD on the MCU over the network.

CiprianRegus, 2023-12-06:
On no-OS, the IIOD network backend expects the `accept()` call to be non-blocking. Can you please make sure if the wiznet accept() implementation behaves in this way?

yyllxxhh, 2023-12-12:
I understand your point; since accept() is non-blocking, when there are no client connection requests, the function will return 
-EAGAIN, allowing the IIOD service to proceed. Thank you for your response.

closed 2023-12-12 by yyllxxhh