analogdevicesinc/no-OS #2117: dmac core cyclic not cleared

dshekter, 2024-03-05:
Hi,

The axi_dmac driver doesn't properly set/clear the cyclic bit in axi_dmac_transfer_start. In my testing, 
```
	// Set flags as needed
	if (dmac->transfer.cyclic != CYCLIC) {
		axi_dmac_read(dmac, AXI_DMAC_REG_FLAGS, &reg_val);
		reg_val = reg_val & ~DMA_CYCLIC;
		axi_dmac_write(dmac, AXI_DMAC_REG_FLAGS, reg_val);
	}
```
added before enabling seems to have worked.

Thanks

danmois, 2024-03-08:
Hello @dshekter ,

Hardware cyclic transfers are possible only for MEM to DEV transfers. The DMA_CYCLIC flag will be set only if the HDL allows cyclic transfers and the transfer size is smaller than the maximum transfer size. Otherwise, the software has to implement the cyclic transfer.

We can continue the conversation on Engineer Zone. If clarifications are needed, please open a thread here: https://ez.analog.com/microcontroller-no-os-drivers/ 

Regards,
George

danmois, 2024-03-11:
Hi @dshekter,

Maybe adding another if statement after the one checking cyclic transfers would do the trick?

if (dmac->transfer.cyclic != CYCLIC)
        reg_val = reg_val & ~DMA_CYCLIC;

We will implement the changes in the following days after testing.

Regards,
George

dshekter, 2024-03-11:
Hi @danmois 

That was initially how I solved it but I would advise against code duplication and instead fix the if statement to operate as you've described its intent. Alternatively, always clear the cyclic bit and then go through the nested if to clear it without the else block. Despite the draw of DMA for systems with relatively capacious memory, I'm trying to keep the code footprint down in my design.

Thank you.
I look forward to the implemented changes, however they are done.

rbolboac, 2024-03-20:
@dshekter , a fix was merged so I am closing this issue.

closed 2024-03-20 by rbolboac