analogdevicesinc/no-OS #2135: Symbolic links break Windows MSDK builds

Brandon-Hurst, 2024-03-28:
The symbolic links for the maxim_dma .c/.h files break builds out-of-the-box on Windows. Windows does not easily import symlinks from Git.

[image]

[image]

Suggested fix: Compile common HAL files via the Makefile includes rather than creating symbolic links.

dbogdan, 2024-04-04:
This should fix the issue: https://github.com/analogdevicesinc/no-OS/commit/b2775b22b47ac1109adf166863bb0a033ef8de92

closed 2024-04-04 by dbogdan

Brandon-Hurst, 2024-04-05:
Hi Dragos,

Thanks, that's great! I noticed that the commit does not make the changes for MAX78000. This MCU should have the same changes done as well.