analogdevicesinc/no-OS #2167: AD7124 driver: bad comparisons

LehMaxence, 2024-04-29:
These two comparisons don't seem to do what's intended:

https://github.com/analogdevicesinc/no-OS/blob/539d8d757eb8eba51d4baddcc817d625b4ecf9fe/drivers/adc/ad7124/ad7124.c#L1041

https://github.com/analogdevicesinc/no-OS/blob/539d8d757eb8eba51d4baddcc817d625b4ecf9fe/drivers/adc/ad7124/ad7124.c#L1044

I saw that when I compiled with IAR which gave me the following warnings:
```console
                if (!(dev->regs[AD7124_ID_REG].value = AD7124_4_ID))
                     ^
"\src\ad7124\ad7124.c",1041  Warning[Pe187]:
          use of "=" where "==" may have been intended

                if (!(dev->regs[AD7124_ID_REG].value = AD7124_8_ID))
                     ^
"\src\ad7124\ad7124.c",1044  Warning[Pe187]:
          use of "=" where "==" may have been intended
```

rbolboac, 2024-04-30:
@D-Disha , looks like you added the lines in https://github.com/analogdevicesinc/no-OS/commit/6d75bc376eb7f1a4fd992cac75a7e33c0c89e37e
can you please fix this?
Thank you!

rbolboac, 2024-05-08:
Hi @LehMaxence ,
Thank you for reporting this.
The fix has been merged with https://github.com/analogdevicesinc/no-OS/pull/2168 .
Let us know if you encounter any other issues.

Ramona

closed 2024-05-08 by rbolboac