analogdevicesinc/no-OS #2172: MAX32690APARD TCP Example IP discrepancy

Brandon-Hurst, 2024-05-02:
The build for the "apard32690" TCP echo server example does not match the functionality described in the documentation OR the Makefile with reference to the IP address that enumerates. Whether or not I define the APARD32690_STATIC_IP macro, the APARD board never responds to the default address 169.254.97.40. Instead, it prints a different address on the serial port on startup:

[image]

Connecting to this port yields a valid TCP echo connection: 
[image]

In addition, the README.md doesn't cover some information that would've been useful in figuring this out: 
- It is not clear from the documentation nor the code that the user should open the serial port to see the IP address of the APARD32690 before they try to connect. Nor is it clear why the IP address should be 169.154.97.37
- The documentation also leaves out how to evaluate this board using Windows tooling (how to add the static IP route mentioned, and also how to ping the APARD32690)

Brandon-Hurst, 2024-05-03:
Figured out the issue and will submit a PR shortly. There's a few things:

- IP, NETMASK, and GATEWAY are not added to the C preprocessor and are needed by lwip_socket.c
- NO_OS_IP, NO_OS_NETMASK, and NO_OS_GATEWAY should have quotes escaped to be evaluated as strings when parsed by the C preprocessor. Example: NO_OS_IP = \"x.x.x.x\"
-  STATIC_IP is disabled by default in the Makefile, but the readme.md suggests the IP address is the static configured address, whereas the default address is different and printed out via serial port after LWIP's DHCP procedure in no_os_lwip_init. Nothing tells the user where to search for the different IP address.

buha, 2024-05-13:
@Brandon-Hurst 
@CiprianRegus fixed this in #2176 , I'm going to close this but feel free to reopen if you have additional questions.

closed 2024-05-13 by buha