analogdevicesinc/no-OS #2179: Zedboard + FMCOMMS3 no-OS ad9361_init : AD936x initialization error

Marvelous-coco, 2024-05-11:
AD9361
hardware:Zedboard + FMCOMMS3 + AD9361

software:vitis2021.1， No-OS 2021.1

I changed nothing in main.c but still failed. The Vitis Serial Terminal reported: 

Calibration TIMEOUT (0x5E, 0x80)
Calibration TIMEOUT (0x244, 0x80)
ad9361_init : AD936x initialization error

----------------------------------------------------------------------------------------

the initiation parameters are:

AD9361_InitParam default_init_param = {
/* Device selection */
ID_AD9361, // dev_sel
/* Reference Clock */
40000000UL, //reference_clk_rate
/* Base Configuration */
0, //two_rx_two_tx_
[image]
mode_enable *** adi,2rx-2tx-mode-enable
1, //one_rx_one_tx_mode_use_rx_num *** adi,1rx-1tx-mode-use-rx-num
1, //one_rx_one_tx_mode_use_tx_num *** adi,1rx-1tx-mode-use-tx-num
1, //frequency_division_duplex_mode_enable *** adi,frequency-division-duplex-mode-enable
0, //frequency_division_duplex_independent_mode_enable *** adi,frequency-division-duplex-independent-mode-enable
0, //tdd_use_dual_synth_mode_enable *** adi,tdd-use-dual-synth-mode-enable
0, //tdd_skip_vco_cal_enable *** adi,tdd-skip-vco-cal-enable
0, //tx_fastlock_delay_ns *** adi,tx-fastlock-delay-ns
0, //rx_fastlock_delay_ns *** adi,rx-fastlock-delay-ns
0, //rx_fastlock_pincontrol_enable *** adi,rx-fastlock-pincontrol-enable
0, //tx_fastlock_pincontrol_enable *** adi,tx-fastlock-pincontrol-enable
0, //external_rx_lo_enable *** adi,external-rx-lo-enable
0, //external_tx_lo_enable *** adi,external-tx-lo-enable
5, //dc_offset_tracking_update_event_mask *** adi,dc-offset-tracking-update-event-mask
6, //dc_offset_attenuation_high_range *** adi,dc-offset-attenuation-high-range
5, //dc_offset_attenuation_low_range *** adi,dc-offset-attenuation-low-range
0x28, //dc_offset_count_high_range *** adi,dc-offset-count-high-range
0x32, //dc_offset_count_low_range *** adi,dc-offset-count-low-range
0, //split_gain_table_mode_enable *** adi,split-gain-table-mode-enable
MAX_SYNTH_FREF, //trx_synthesizer_target_fref_overwrite_hz *** adi,trx-synthesizer-target-fref-overwrite-hz
0, // qec_tracking_slow_mode_enable *** adi,qec-tracking-slow-mode-enable
/* ENSM Control */
0, //ensm_enable_pin_pulse_mode_enable *** adi,ensm-enable-pin-pulse-mode-enable
0, //ensm_enable_txnrx_control_enable *** adi,ensm-enable-txnrx-control-enable
/* LO Control */
2400000000UL, //rx_synthesizer_frequency_hz *** adi,rx-synthesizer-frequency-hz
2400000000UL, //tx_synthesizer_frequency_hz *** adi,tx-synthesizer-frequency-hz
1, //tx_lo_powerdown_managed_enable *** adi,tx-lo-powerdown-managed-enable
/* Rate & BW Control */
{983040000, 245760000, 122880000, 61440000, 30720000, 30720000},// rx_path_clock_frequencies[6] *** adi,rx-path-clock-frequencies
{983040000, 122880000, 122880000, 61440000, 30720000, 30720000},// tx_path_clock_frequencies[6] *** adi,tx-path-clock-frequencies
18000000,//rf_rx_bandwidth_hz *** adi,rf-rx-bandwidth-hz
18000000,//rf_tx_bandwidth_hz *** adi,rf-tx-bandwidth-hz
/* RF Port Control */

I just want to receive I/Q signal through AD9361.

Could you please help me? or somebuddy occured same question?

Thank you!

dbogdan, 2024-05-13:
Hi @Marvelous-coco,
Can you please specify the exact name of the HDL and no-OS branches? Do you have the option of using the main branches for both HDL and no-OS?
Thanks,
Dragos

Marvelous-coco, 2024-05-13:
> Hi @Marvelous-coco, Can you please specify the exact name of the HDL and no-OS branches? Do you have the option of using the main branches for both HDL and no-OS? Thanks, Dragos
HDL branch：hdl-hdl_2021_r1
no-OS branch：no-OS-2021_R1
vitis&vivado: 2021.1
I followed the [NO-OS Build Guide](https://wiki.analog.com/resources/no-os/build) to build no-OS main branch->projects->AD9361, but I failed to build it. Then I tried to build no-OS-2021_R1 branch and successfully built the project. But now I can't run the AD9361 project because of the initialization error .  Do you mean it is something wrong in building HDL project?
Thank you for you reply!

Ravioora, 2024-05-18:
> ****

buha, 2024-10-02:
Hi, so the currently supported releases are 2022 r1 and r2 and of course the latest `main` branch. If you have errors building on these branches we're happy to help. Sorry for reply delay, feel free to reopen if you are working on this and need support, preferably on EngineerZone forum.

closed 2024-10-02 by buha