analogdevicesinc/no-OS #2283: AD7606C-16 Device ID is incorrect

AdCalleja, 2024-08-26:
The datasheet value for the Device ID and Silicon Revision is wrong, and thus, the [chip info of the driver](https://github.com/analogdevicesinc/no-OS/blob/2c4d7b8ec3fb4af4bf3ce430d5e2dc2777e143b6/drivers/adc/ad7606/ad7606.c#L166). 
There is a FAQ article that gives the correct values of 0x22: [What's the ID register content for each generic?](https://ez.analog.com/data_converters/precision_adcs/w/documents/15710/what-s-the-id-register-content-for-each-generic).

The proposed solution is to update [.device_id](https://github.com/analogdevicesinc/no-OS/blob/2c4d7b8ec3fb4af4bf3ce430d5e2dc2777e143b6/drivers/adc/ad7606/ad7606.c#L173) to 0x2, referencing the FAQ to notify about the error in the datasheet.

If it's within someone's reach, it would be helpful to raise this error to have an updated datasheet.

amiclaus, 2024-09-16:
Hello @AdCalleja, 
Did you test with actual hardware and encountered this issue for AD7606C-16 such that the device_id is 0x2 when reading it from the device?
Regards,

AdCalleja, 2024-09-16:
Correct, I encountered this behaviour while testing actual hardware, and then found the FAQ note.

amiclaus, 2024-09-17:
Hello,

I added a patch for this issue: https://github.com/analogdevicesinc/no-OS/pull/2298

Regards,

closed 2024-09-23 by amiclaus