analogdevicesinc/no-OS #2431: ADXL355 adxl355_get_raw_fifo_data() random issue

frastrello, 2025-02-03:
Hi,

after some vibrational/smal shock tests, I started to get randomly (it happnes once in 5/10 minutes) a vector of `raw_x` with 10-15 random wrong values completely out of range, but it happens only for x values, not for y and z values. I checked that, when it happens, the function returns a number of `fifo_entries ` that is not a multiple of 3 and it seems strange to me... how could it be the root cause? Could the ADXL355 be damaged?

Thanks,
Fabio

CiprianRegus, 2025-02-03:
Can you please give us more details about your hardware setup? What platform are you testing is on? Are you using the sample project from this repository?

frastrello, 2025-02-06:
> Can you please give us more details about your hardware setup? What platform are you testing is on? Are you using the sample project from this repository?

Hi CiprianRegus, thanks for the feedback.

Hereafter the technical details:
- I'm using a STM32U5A5 with SPI connection at 4MHz to acquire data from the ADXL355
- I'm using FreeRTOS in my project but I successfully integrted the NO_OS library 
- I'm using the FIFO acquisition with a read of the fifo buffers in a dedicated high priority task after capturing the FIFO FULL interrupt
- I'm also continuously monitoring the FIFO OVERRUN interrupt i vonfigured to check if i loose  samples, and i don't loose samples

I started to test with the ADXL355Z platform and NUCLEO-U5A5ZJ with success, now I'm testing on custom boards.
Moreover, I'm not using your sample project from your repoisitory.

buha, 2025-02-06:
SPI issues would be visible on all channels so not the case.

I was also about to ask you to check the overrun condition, if it ever happens but you already mentioned it doesn't so we can eliminate that.

Could you try to put an else statement to this if ?
https://github.com/analogdevicesinc/no-OS/blob/ac2ed6cebc99d2d94fe188502cb6c0dd6a883a3b/drivers/accel/adxl355/adxl355.c#L742

and an `asm("nop;");` under the else branch, a breakpoint on it and see if it ever hits in between two such instances where data is corrupt for X axis

One other thing I'd try is see how it performs by adding `*fifo_entries = (*fifo_entries / 3) * 3` right after reading it, to force it to read only a multiple of 3 in the event that the FIFO gets filled sequentially with raw data from channels and you happened to nail the moment when it's getting filled and the total count is not a multiple of 3 (yet). It seems from datasheet that sampling is simultaneous while FIFO loading is sequential although I'm not 100% sure of how it works internally.

frastrello, 2025-02-06:
Hi [buha](https://github.com/buha), thanks for the suggestion.

I already monitored in a similar way, this is what I get:
 [image]

I already added  this:
```
	uint8_t leftover_new = *fifo_entries % 3;
	if(leftover_new !=0){
		*fifo_entries = *fifo_entries - leftover_new;
	}
```
but only outside the library, in my code, to avoid to touch the library, but I will try to do it in a lower level as you suggest.

buha, 2025-02-07:
My expectation is that you are not losing, because you are reading data only as multiples of 3 so whatever is left there in the fifo by not reading it, will stay there and eventually be read in the future when new samples get pushed into the fifo. So if no overflow occurs, you should not lose it.

Yes, it's a bug in `adxl355_get_nb_of_fifo_entries()`, I will modify the code.

commit 8c3dbcf2 (2025-02-07): adxl355 - read FIFO as multiples of 3

Data gets stored in the FIFO as locations and 3 locations pertain
to the same measurement (x, y, z). However, it may be that one
interrogates the number of filled locations in the FIFO and this
number is not a multiple of 3 at that moment (most likely due
to locations being filled sequentially). In this case, the correct
approach is to read only multiples of 3 and leave the remaining 1
or 2 locations in the FIFO. This commit does that.

Closes #2431
closed 2025-02-07 by buha