analogdevicesinc/no-OS #2487: Vitis Project Not Created, Missing Necessary Files for Running Application

MehmettAkiff, 2025-03-04:
I followed the instructions on the No-OS GitHub Wiki (https://wiki.analog.com/resources/no-os/build), but I wasn't able to create a runnable application in the Vitis environment. Specifically, after using the make command to generate the necessary files, I ran the make sdkopen command, but it only opened a blank Vitis window. No Vitis project was created. (src , build , .Xil folders are created.)

I don't know how to manually add these files, and I need a fully configured Vitis project that works out of the box. Could you provide me with a Vitis project that includes the necessary files and is ready to run in the Vitis environment?

Thank you.

tfcollins, 2025-03-04:
What project/hardware are you trying to build?

buha, 2025-03-05:
There is a nasty Vitis behavior that you need to know. When you use `make sdkopen` after generating the project with our make system, click X on the "Welcome" tab in Vitis to close it, this will reveal the project and all the files on the left pane.

closed 2025-03-07 by buha