analogdevicesinc/no-OS #2516: AD5940 driver ad5940_SweepNext function is skipping first frequency point

phryniszak, 2025-03-28:
As in topic.

If you enable frequencies sweep in one of examples like this:

```
	.SweepCfg.SweepEn = true, // true false
	.SweepCfg.SweepStart = 10000,
	.SweepCfg.SweepStop = 80000.0,
	.SweepCfg.SweepPoints = 4,
	.SweepCfg.SweepLog = false,
	.SweepCfg.SweepIndex = 0,

```

first sweep gives you frequencies without first point 10000:

[00:00:06.375,427] <inf> ph_ad594x: Freq:33333.33
[00:00:11.840,026] <inf> ph_ad594x: Freq:56666.67
[00:00:11.891,632] <inf> ph_ad594x: Freq:80000.00

and following run correct values:

[00:00:11.943,206] <inf> ph_ad594x: Freq:10000.00
[00:00:13.402,465] <inf> ph_ad594x: Freq:33333.33
[00:00:13.454,101] <inf> ph_ad594x: Freq:56666.67
[00:00:13.505,645] <inf> ph_ad594x: Freq:80000.00

most likely it is because of this line:

			if (++pSweepCfg->SweepIndex == pSweepCfg->SweepPoints)

buha, 2025-03-31:
yes, i reviewed the code and that's a bug, the ++ beforehand skips the first point, could you try this version ? 

```
void ad5940_SweepNext(struct ad5940_dev *dev, SoftSweepCfg_Type *pSweepCfg,
		      float *pNextFreq)
{
	float frequency;

	if (pSweepCfg->SweepPoints < 2)
		return -EINVAL;

	if (pSweepCfg->SweepStart < 0 || pSweepCfg->SweepStop < 0)
		return -EINVAL;

	int delta = (pSweepCfg->SweepStart < pSweepCfg->SweepStop) ? 1 : -1;

	if (pSweepCfg->SweepLog) { /* Log step */
		if (delta > 0) {
			frequency = pSweepCfg->SweepStart * pow(10,
								pSweepCfg->SweepIndex * log10(pSweepCfg->SweepStop / pSweepCfg->SweepStart) /
								(pSweepCfg->SweepPoints - 1));
			pSweepCfg->SweepIndex++;
			if (pSweepCfg->SweepIndex == pSweepCfg->SweepPoints)
				pSweepCfg->SweepIndex = 0;
		} else {
			frequency = pSweepCfg->SweepStart * pow(10, pSweepCfg->SweepIndex *
								(log10(pSweepCfg->SweepStart / pSweepCfg->SweepStop) / (pSweepCfg->SweepPoints -
										1)));
			pSweepCfg->SweepIndex--;
			if (pSweepCfg->SweepIndex == 0) {
				pSweepCfg->SweepIndex = pSweepCfg->SweepPoints - 1;
			}
		}
	} else { /* Linear step */
		frequency = pSweepCfg->SweepStart + (double)pSweepCfg->SweepIndex *
			    (pSweepCfg->SweepStop - pSweepCfg->SweepStart) / (pSweepCfg->SweepPoints - 1);
		pSweepCfg->SweepIndex++;
		if (pSweepCfg->SweepIndex == pSweepCfg->SweepPoints) {
			pSweepCfg->SweepIndex = 0;
		}
	}

	*pNextFreq = frequency;
}
```

phryniszak, 2025-03-31:
But now we have to change function signature as you want to return -EINVAL

```
	if (pSweepCfg->SweepPoints < 2)
		return -EINVAL;

	if (pSweepCfg->SweepStart < 0 || pSweepCfg->SweepStop < 0)
		return -EINVAL;
```

phryniszak, 2025-03-31:
I didn't analyze just run copilot and this what I get and it seems to be working:

```
void ad5940_SweepNext(struct ad5940_dev *dev, SoftSweepCfg_Type *pSweepCfg,
                      float *pNextFreq)
{
    float frequency;

    if (pSweepCfg->SweepLog)
    { /* Log step */
        if (pSweepCfg->SweepStart < pSweepCfg->SweepStop)
        { /* Normal */
            frequency = pSweepCfg->SweepStart * pow(10,
                                                    pSweepCfg->SweepIndex * log10(pSweepCfg->SweepStop / pSweepCfg->SweepStart) /
                                                        (pSweepCfg->SweepPoints - 1));
            if (++pSweepCfg->SweepIndex == pSweepCfg->SweepPoints)
                pSweepCfg->SweepIndex = 0;
        }
        else
        {
            frequency = pSweepCfg->SweepStop * pow(10, pSweepCfg->SweepIndex *
                                                           (log10(pSweepCfg->SweepStart / pSweepCfg->SweepStop) / (pSweepCfg->SweepPoints -
                                                                                                                   1)));
            if (pSweepCfg->SweepIndex-- == 0)
                pSweepCfg->SweepIndex = pSweepCfg->SweepPoints - 1;
        }
    }
    else
    { /* Linear step */
        if (pSweepCfg->SweepStart < pSweepCfg->SweepStop)
        { /* Normal */
            frequency = pSweepCfg->SweepStart + pSweepCfg->SweepIndex * (double)(pSweepCfg->SweepStop - pSweepCfg->SweepStart) / (pSweepCfg->SweepPoints - 1);
            if (++pSweepCfg->SweepIndex == pSweepCfg->SweepPoints)
                pSweepCfg->SweepIndex = 0;
        }
        else
        {
            frequency = pSweepCfg->SweepStop + pSweepCfg->SweepIndex * (double)(pSweepCfg->SweepStart - pSweepCfg->SweepStop) / (pSweepCfg->SweepPoints - 1);
            if (pSweepCfg->SweepIndex-- == 0)
                pSweepCfg->SweepIndex = pSweepCfg->SweepPoints - 1;
        }
    }

    *pNextFreq = frequency;
}
```

buha, 2025-04-01:
yeah i missed that the function returns void there (shouldn't, to be honest, with so many failure conditions... )

i edited my comment, i just wanna know if it works for you as it is now, as i don't have hardware to test this on right now

phryniszak, 2025-04-01:
It is fine now:

```
00:00:02.443,389] <inf> ph_ad594x: Freq:10000.00
[00:00:02.495,178] <inf> ph_ad594x: Freq:33333.33
[00:00:02.546,936] <inf> ph_ad594x: Freq:56666.67
[00:00:02.598,693] <inf> ph_ad594x: Freq:80000.00

```
 I also agree with you that silent failing in edge cases is not the best strategy.