analogdevicesinc/no-OS #2518: Linux I2C Driver unable to read register in adt7420

jmacagba, 2025-04-01:
In both [linux_i2c_read](https://github.com/analogdevicesinc/no-OS/blob/ed20ea5cf06578b2139035b4b9243d397a4e49cb/drivers/platform/linux/linux_i2c.c#L176) and [linux_i2c_write](https://github.com/analogdevicesinc/no-OS/blob/ed20ea5cf06578b2139035b4b9243d397a4e49cb/drivers/platform/linux/linux_i2c.c#L137), the **stop_bit** condition is ignored;
- However, as per my understanding, this is crucial for I2C communication, since we might need a repeated start condition on certain use cases.

The error I have encountered is when reading a register for adt7420 which is always 0. 

[image]

- From what I understand,
  - LOC 453 will write the register address (that needs to be read) and a **stop_bit** should not be generated (_However, when using Linux Platform, stop_bit does not do anything_)
  - LOC 456, will read back the register value and a **stop_bit** should be generated (_However, when using Linux Platform, stop_bit does not do anything_)
- I think the issue is with using `write()` and `read()` for I2C, which cannot use/implement **stop_bit** (where in each write/read might automatically send a **stop_bit**)
  - _Note: I was able to replicate this issue in a simple C Program_
- Instead, I have found that using `ioctl(file,I2C_RDWR, &packets);` or `__s32 i2c_smbus_read_byte_data(int file, __u8 command);` are working alternatives
  - _Note: I was only able to test this in a simple C program and have not tested it on the Linux I2C Driver._
  - Reference: [Implementing I2C device drivers in userspace — The Linux Kernel documentation](https://www.kernel.org/doc/html/latest/i2c/dev-interface.html)

buha, 2025-04-04:
Hi, I checked the current implementation of our I2C API for the linux userspace platform, and indeed, the `stop_bit` is ignored. 

https://github.com/analogdevicesinc/no-OS/blob/main/drivers/platform/linux/linux_i2c.c#L160

This is definitely something that needs to be fixed, we will do it but I can't offer a timeline. If you would like to contribute and make a pull request that would be awesome!

jmacagba, 2025-05-07:
Hi @buha ,

Me and @arthur428 looked into this and will do a PR.

arthur428, 2025-05-08:
Hi @buha,

Here is our PR: https://github.com/analogdevicesinc/no-OS/pull/2593

Let us know if there is anything we should change. Thanks.

jmacagba, 2025-06-20:
Hi @buha, since the PR https://github.com/analogdevicesinc/no-OS/pull/2593 has already been merged, can this issue be closed?