analogdevicesinc/no-OS #2600: ADRV9002 NO OS

hyanki, 2025-05-13:
Hi 
i want to use no os driver for my application. Build instruction is given and able to generate bootables but
How can i debug the code while developing application?
Regards
Anki

buha, 2025-05-20:
Use the `make sdkopen` command, which opens up the IDE (Vitis), close the Welcome page and build+debug the project like you'd usually do in Vitis by clicking the debugger (bug) button.

hyanki, 2025-05-20:
Hi buha
I want to port in altera SOC, So IDE i want is ARMDS5 altera version. Please give steps.
Regards
Anki

buha, 2025-05-22:
I am not familiar with it and we do not provide any examples that can be build with that toolchain.

hyanki, 2025-05-22:
Ok buha. Thanks for the reply.

Anki